rm-hull/OPi.GPIO #3: Device or resource busy when cleanup() is not used

user2684, 2017-03-28:
Hi, really an excellent job @rm-hull first of all!

Not sure if it is expected but when I configure a pin as output and forget to use GPIO.cleanup() at the end of my script, it is no more possible to change the value of the output. E.g. if I run:

import OPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
pin = 11
GPIO.setup(pin, GPIO.OUT)
GPIO.output(pin, GPIO.HIGH)

And then:

import OPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
pin = 11
GPIO.setup(pin, GPIO.OUT)
GPIO.output(pin, GPIO.LOW)

I get the following error: 

Traceback (most recent call last):
  File "gpio.py", line 4, in <module>
    GPIO.setup(pin, GPIO.OUT)
  File "/usr/local/lib/python2.7/dist-packages/OPi/GPIO.py", line 241, in setup
    sysfs.export(pin)
  File "/usr/local/lib/python2.7/dist-packages/OPi/sysfs.py", line 20, in export
    fp.write(str(pin))
IOError: [Errno 16] Device or resource busy

I guess because GPIO.setup tries to export the pin again causing the error and preventing the program to continue running. Not using setup() or catching the error results in a Channel is not configured  message.
Thanks

rm-hull, 2017-03-28:
Feedback like this is really useful, so thanks.. and I'm glad you are finding it useful !

If you look at https://github.com/rm-hull/OPi.GPIO/blame/master/OPi/GPIO.py#L240-L247 you can see that it traps `OSError` and checks for errno == 16, and tries to unexport and then attempts to re-export.

Perhaps it ought to also check for `IOError` as well.
I will fix it up and push out a new release shortly

closed 2017-03-28 by rm-hull

rm-hull, 2017-03-28:
Version 0.2.2 released which _should_ attempt to re-export the pin if it is already in use (by default it will log a warning, but you can switch the warnings off). 

Ideally though, your script should call cleanup on exit.

Re-install with: `sudo pip install --upgrade OPi.GPIO`

user2684, 2017-03-28:
Excellent, thanks for the quick fix, I can confirm it is working fine now, I see the warning as you mentioned but does not halt anymore. For your information, I'm using your library in the dev branch of my project https://sourceforge.net/projects/my-house

commit 14465ad9 (2017-03-28): Also catch IOError (which also may get thrown and raise errno==16)

Fixes #3
PR #4 (2017-03-28): Also catch IOError (which also may get thrown and raise errno==16)
Fixes #3
commit db91dc38 (2017-03-28): Also catch IOError (which also may get thrown and raise errno==16) (#4)

Fixes #3