rm-hull/OPi.GPIO #5: add_event_detect() prevent the script to exit

user2684, 2017-03-28:
Not sure this is related to the library but just in case...
When I run the test script below, it works fine but then when I press ctrl+c, it does not exit. 

```python
import OPi.GPIO as GPIO
import time
GPIO.setmode(GPIO.BCM)
pin = 9
GPIO.setup(pin, GPIO.IN,initial=GPIO.HIGH)
def event_callback(pin):
        print pin
        print GPIO.input(pin)

GPIO.add_event_detect(pin, GPIO.FALLING, callback=event_callback)
while True:
        time.sleep(10)
```

Looks like the interrupt is caught hence sleep() is interrupted but then the only way to close the script is to kill it from a different console.
```
Traceback (most recent call last):
  File "gpio.py", line 12, in <module>
    time.sleep(10)
KeyboardInterrupt
```
If I comment out GPIO.add_event_detect(pin, GPIO.FALLING, callback=event_callback), it exists normally.
Thanks

rm-hull, 2017-03-28:
Definitely sounds like a bug in the library, thanks for the report. Will look into it shortly.

closed 2017-05-26 by rm-hull

rm-hull, 2017-05-26:
0.2.3 has just been released which should fix the issue - I tried your script and it now exits when Ctrl+C pressed. Let me know if you run into problems

user2684, 2017-05-27:
Great, I'll give it a try in the next few days and let you know

commit cf76f06e (2017-05-26): Make worker thread daemonic - fixes #5 (#7)