rm-hull/OPi.GPIO #6: Support for Pull up / Pull down resistors

user2684, 2017-03-28:
It would be great (if not too complex) to add support for the RPi.GPIO `pull_up_down` argument of setup() since it is very common to have pull up/down set when reading an input. 
I've tried the trick to output e.g. HIGH before reading but I correctly got an error saying the pin is supposed to be an input not an output.
Alternatively, I'd recommend adding the pull_up_down option anyway and just print out a warning saying it is not implemented so to ensure compatibility with existing code. A similar approach can be used for bouncetime and other optional parameters in this way the code will be fully compatible even if the functionality is not implemented.

Thanks

rm-hull, 2017-03-28:
Yes, pull-up and pull-down arguments are on the radar, but at present I'm not sure if it is fully supported by the GPIOs on the Orange Pi Zero. But I like your idea of accepting them regardless and displaying a warning.

Bouncetime should be easier to support, but in the meantime I'll also add the params and emit some warnings.

closed 2017-05-27 by rm-hull

rm-hull, 2017-05-27:
Version 0.2.4 adds compatibility (but no functionality other than an emitted warning) for pull-up/down resistors and bouncetime.

user2684, 2017-05-27:
Good job, this will for sure help me in making the code cleaner without too many if based on the platform :-) (https://sourceforge.net/p/my-house/tickets/216/)

PR #8 (2017-05-27): Add compatibility for pull up/down and bouncetime params
Prints a warning, as the actual functionality is still missing, but existing code will not break.

Fixes #6