rm-hull/OPi.GPIO #10: Support for bouncetime

rm-hull, 2017-05-27:
Pull request #8 added faux-compatibility for bouncetime by adding an optional parameter, so that existing code doesn't break. This issue is a placeholder to add real support for bouncetime.

Related (originating) issue: #6

shaunmulligan, 2017-07-30:
Hey @rm-hull thanks for this great library, I am currently trying to use it for a simple user interface and hitting into the problem of multiple edge events per button press. Do you have any recommended work around or tips/pointers on implementing this feature?

gw0, 2017-11-19:
Isn't the solution to just remember the time at millisecond resolution of the last edge to which the library responded and then ignore all edges until `bouncetime` time has passed?

adonespitogo, 2020-01-18:
Is there a work in progress on this? How can we contribute for this?

TomW1605, 2022-12-01:
hi, is there any progress on this one? also happy to take a shot at it if that would help

rm-hull, 2022-12-01:
Go for it. I'm not active on this project but will merge PRs and publish to pypi if folks want to submit changed