rm-hull/OPi.GPIO #9: Support for Pull up / Pull down resistors

rm-hull, 2017-05-27:
Pull request #8 added faux-compatibility for pull up/down resistors by adding an optional parameter, so that existing code doesn't break. This issue is a placeholder to add real support for pull-up/down resistors.

Related (originating) issue: #6

gw0, 2017-11-19:
As I understand, the `sysfs` interface does not support dynamic configuration of internal pull-up/down resistors. So an Allwinner-specific solution is necessary, just like Raspberry Pi has Broadcom-specific solutions.

- The Allwinner-specific `sunxi` interface supports this, but you have to access the system memory directly. Is it compatible with `sysfs` interface?
- Another option seems to be writing DTS overlays ([like on RPi](https://stackoverflow.com/questions/33626480/raspberry-pi-gpio-pull-up-down-resistors-with-sysfs)) and rebooting each time. But it seems this is not yet supported (`/boot/dtb-4.13.8-sunxi/overlay/README.sun8i-h3-overlays`)?
- A similar Allwinner-specific solution are Fex Guide [GPIO configuration](http://linux-sunxi.org/Fex_Guide#Port_Definitions) and rebooting each time?

enesbcs, 2020-04-15:
I've found a DTS that enables internal pullup, not sure if it can be used.
https://github.com/armbian/sunxi-DT-overlays/blob/master/examples/gpio-button.dts