rm-hull/OPi.GPIO #11: IOError when trying to setup PA10 pin

mtanwr, 2017-06-11:
Hi,

I am trying to setup the following PA10 pin as input in the "BOARD" mode. 

Other references to this pin:
Orange Pi = SIM_DET/PA_EINT10/PA10
Physical Pin = 26

My code:

import OPi.GPIO as GPIO
GPIO.setmode(GPIO.BOARD)
GPIO.setup(26, GPIO.IN)

Error message:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/OPi/GPIO.py", line 356, in setup
    sysfs.unexport(pin)
  File "/usr/local/lib/python2.7/dist-packages/OPi/sysfs.py", line 26, in unexport
    fp.write(str(pin))
IOError: [Errno 22] Invalid argument

I don't get this error in any other pins when I try to set them as input. Am I missing something?

rm-hull, 2017-06-11:
A quick google brought up this page: https://forum.armbian.com/index.php?/topic/3178-orange-pi-zero-pa10-is-blocked/ which could account for the behaviour you describe... Maybe try disabling the audio config as per the replies?

Let me know how it goes.

mtanwr, 2017-06-11:
@rm-hull thanks for the link. I'll take a look and get back to you with the results.

mtanwr, 2017-06-12:
@rm-hull 

So the issue described in that link wasn't exactly the same in my case. But it lead me to something similar. In my OPi, the PA10 was assigned for 1-wire components. I checked /sys/kernel/debug/gpio which showed:

```
GPIOs 0-383, platform/sunxi-pinctrl, sunxi-pinctrl:
 gpio-10  (w1               )         in  hi
 gpio-17  (red_led             )    out lo
 gpio-69  (?                   )        out hi
 gpio-202 (xradio_irq          )  in  lo
 gpio-354 (?                   )       out hi
 gpio-362 (green_led           ) out hi
```

You can see the w1 next to GPIO-10. After that I converted /boot/bin/orangepizero.bin to fex format, and found the following lines:

```
[w1_para]
w1_used = 1
gpio = 10
```

I changed it to the following, saved and replaced the old bin file and restarted the OPi

```
[w1_para]
w1_used = 0
gpio =
```

Channel 26/PA10 can now be setup as an input with no problems.

rm-hull, 2017-06-12:
Excellent bit of sleuthing !

closed 2017-06-12 by rm-hull