rm-hull/OPi.GPIO #12: Pin Numbering

ThisIsQasim, 2017-06-25:
Could you please elaborate more on the sunxi numbering system. Thank you

rm-hull, 2017-06-25:
Hello, it is just (yet another) numbering system that can be used to reference the pins.

[image]

If you are specifically targeting an Orange Pi rather than a Raspberry Pi it might make more sense to, e.g., refer to PA11 and PA12 for the I2C data & clock lines

rm-hull, 2017-07-15:
Please re-open if you need any more assistance

closed 2017-07-15 by rm-hull

ThisIsQasim, 2017-07-16:
I am using the BCM numbers for now but If you could provide a sample code that would help. The confusing part is that the PIN numbers are strings so can we pass these strings as arguments to the standard functions?