rm-hull/OPi.GPIO #14: race condition when using the library as non root user

olivluca, 2017-09-25:
I have an udev rule in place that allows me to use the GPIOs as a normal user

    SUBSYSTEM=="gpio", ACTION=="add", RUN="/bin/sh -c '\
      chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
      chown -R root:gpio /sys$DEVPATH && chmod -R 770 /sys$DEVPATH"

The problem is that, after writing to the /sys/class/gpio/export files, udev takes a little time to adjust the permissions, so when the library accesses the gpio file it doesn't have the permission to do so.
As a workaround I added a delay in GPIO.py (see attached patch [opi.zip](https://github.com/rm-hull/OPi.GPIO/files/1330195/opi.zip)) but I think there should be a better way .
See also https://forum.armbian.com/index.php?/topic/5225-gpio-from-python-user-space-opigpio-app/

rm-hull, 2017-10-03:
Adding a sleep into the code doesn't seem like the right approach here - how do you know if 100ms is long enough for all cases?

I would expect that what _should_ happen is that the kernel devices get created with a group id of `gpio` in the first place and the user is also assigned that group. You would need to appeal to the Armbian maintainers to do this I suppose though: but doing that 'fixes' it for all consumers of the gpio subsystem then, not just OPI.GPIO.

olivluca, 2017-10-03:
No, the device gets created with root.root and permission 600, it's the udev script that it changes it afterwards and apparently it does it asynchronously. I don't know if 100ms is enough, that's why I said there should be a better way. Maybe check if the file is accessible in a loop and give up after a timeout.

olivluca, 2017-10-03:
s/accessible/writeable/

rm-hull, 2018-02-22:
I have proposed a solution in the comments in https://github.com/rm-hull/OPi.GPIO/pull/25#pullrequestreview-98420322 - please review and comment if you think this would be an acceptable compromise to the suggestion here, in #15 and #23

cc @olivluca, @philfifi, @elpimous, @sgjava

olivluca, 2018-02-22:
Both the delay and the proposed callback are just workarounds: I still think the proper solution would be to check for the accessibility of the device file, and the proposed callback has no knowledge of that.