rm-hull/OPi.GPIO #16: Pin mapping feature request

sgjava, 2017-12-30:
I like the idea of adding SBC based mappings, but I would also like the BCM mappings. What I started with was adding dicts BOARD_DUO and BCM_DUO dicts. What do you think? Also, does it make sense to call the project OPi when it's really sysfs.GPIO? Let me know it you like the dicts idea and I'll do a PR for Duo. There's also the Duo Mini Shield, so I'm going to see if I need another set of dicts or if I can just use the Duo mappings.

rm-hull, 2017-12-30:
How about (externally to the project, at this point) a module called, say, _neo/duo.py_ that contains:

```python
BOARD = {
    3: 12,   # I2C0_SDA/GPIOA12
    5: 11,   # I2C0_SCL/GPIOA11
    7: 203,  # GPIOG11
    8: 198,  # UART1_TX/GPIOG6
    10: 199, # UART1_RX/GPIOG7 
    11: 5,   # DEBUG_RX(UART_RXD0)/GPIOA5/PWM0
    12: 363, # GPIOL11/IR-RX
    13: 4,   # DEBUG_TX(UART_TXD0)/GPIOA4 
    19: 15,  # UART3_RTS/SPI1_MOSI/GPIOA15
    21: 16,  # UART3_CTS/SPI1_MISO/GPIOA16
    23: 14,  # UART3_RX/SPI1_CLK/GPIOA14
    24: 13,  # UART3_TX/SPI1_CS/GPIOA13
    41: 355  # Button
}

# NanoPi Duo BCM pin to actual GPIO pin
BCM = {
    2: 12,   # I2C0_SDA/GPIOA12
    3: 11,   # I2C0_SCL/GPIOA11
    4: 203,  # GPIOG11
    8: 13,   # UART3_TX/SPI1_CS/GPIOA13
    9: 16,   # UART3_CTS/SPI1_MISO/GPIOA16
    10: 15,  # UART3_RTS/SPI1_MOSI/GPIOA15
    11: 14,  # UART3_RX/SPI1_CLK/GPIOA14
    14: 198, # UART1_TX/GPIOG6
    15: 199, # UART1_RX/GPIOG7
    17: 5,   # DEBUG_RX(UART_RXD0)/GPIOA5/PWM0
    18: 363, # GPIOL11/IR-RX
    27: 4,   # DEBUG_TX(UART_TXD0)/GPIOA4
    41: 355  # Button
}
```

Then to initialize with this:

```python
import OPi.GPIO as GPIO
import neo.duo

GPIO.setmode(neo.duo.BCM)  # or GPIO.setmode(neo.duo.BOARD)
GPIO.setup(15, ...)  # <-- 199, # UART1_RX/GPIOG7
```

`setmode` would need to change to detect a dict in the first argument, and internally sets a custom mode from the supplied pin map.

Over time, we could then move out the OPi specific mappings either externally (e.g `GPIO.setMode(orangepi.zero.BCM`), else bring in nanopi mappings internally into the project.

As to renaming to *sysfs.gpio* or *SBC.GPIO* (too shouty maybe?), definitely, but that's going to need a shakeup of the documentation as well and ought be tackled as a separate endeavor.

sgjava, 2017-12-30:
This looks good to me. Do you want to make the changes and toss my PR?

rm-hull, 2017-12-30:
Will do 👍

sgjava, 2017-12-30:
Let me know when I can test and I will report results, thanks!

rm-hull, 2017-12-30:
There's a branch with some initial code changes: [feature/alt-pin-mappings](https://github.com/rm-hull/OPi.GPIO/tree/feature/alt-pin-mappings) - bear in mind it is untested at the moment!

sgjava, 2017-12-30:
Will this work with the examples, i.e. --gpio GPIO? or will you need to add another command line parameter?

rm-hull, 2017-12-31:
Probably not, in which case we would need another param like `--gpio-mode=neo.duo.BCM`.
A simple hard-coded test harness would work though

sgjava, 2017-12-31:
No worries, I'll test that as well. Right now I'm writing up how to configure Luma for the Duo and I2C display on the Armbian forum. Once that is done I'll jump on this, thanks.

rm-hull, 2017-12-31:
There's a PR here: https://github.com/rm-hull/OPi.GPIO/pull/18 - tests all pass, so I think we could make a release (but will be tomorrow - it's late here now)

sgjava, 2017-12-31:
Good, will give me time to test. I'll report back. As soon as I'm done documenting the Duo I2C stuff I'll hit the SPI/GPIO stuff next. Have a good night.

rm-hull, 2017-12-31:
Version 0.3.0 released to support this functionality

sgjava, 2017-12-31:
I'm going to fork and add module as we've been discussing. If that works should I create a PR?

sgjava, 2017-12-31:
OK, fork works like a charm:

```
import time
import nanopi.duo
from OPi import GPIO
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.oled.device import ssd1306

GPIO.setmode(nanopi.duo.BOARD)
GPIO.setwarnings(False)
serial = spi(port=1, device=0, gpio=GPIO, gpio_DC=5, gpio_RST=3)
device = ssd1306(serial)

with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white")
    draw.text((3, 3), "Hello", fill="white")

time.sleep(10)
```

rm-hull, 2017-12-31:
re. pins left open - when the GPIO is supplied to luma components, it is treated as 'unmanaged' and thus it is up to the caller to clean up. This is by design, see: 
* https://github.com/rm-hull/luma.core/blob/master/luma/core/interface/serial.py#L147
* https://github.com/rm-hull/luma.core/blob/master/luma/core/interface/serial.py#L212-L217

sgjava, 2017-12-31:
OK, I probably didn't notice before due to GPIO.setwarnings(False). So, should I try to test the examples with a new pin mappings param?

rm-hull, 2017-12-31:
> So, should I try to test the examples with a new pin mappings param?

Go for it - I will try and add the `--gpio-mode` param to luma.core as discussed but might not be for a few days

> I'm going to fork and add module as we've been discussing. If that works should I create a PR?

Yes please - the only problem we might encounter here is that when it packages things up, if we introduce a new top-level namespace, then we'll have to add it [here](https://github.com/rm-hull/OPi.GPIO/blob/master/setup.py#L34) as well.

sgjava, 2017-12-31:
OK, I added nanopi package to setup.py, did a pull and tested install again. All is good there. After you add the new cmdline param I'll test it.

sgjava, 2017-12-31:
Test luma.examples by creating local copy of cmdline, changing demo_opts import, changing GPIO.setmode and it worked perfect! If you want I can try my hand at adding new param to cmdline and creating PR? I'm going to create the PR for OPi.GPIO now.

rm-hull, 2017-12-31:
> If you want I can try my hand at adding new param to cmdline and creating PR?

Absolutely, go for it!

rm-hull, 2018-01-01:
@sgjava - see https://github.com/rm-hull/luma.core/pull/117 (work-in-progress) for addition of `--gpio-mode` flag

sgjava, 2018-01-01:
OK, we put this to bed, thanks. :smiley: I'll look at the cmdline thing next.

closed 2018-01-01 by sgjava