rm-hull/OPi.GPIO #21: sysfs deprecated

sgjava, 2018-01-01:
See my post on Armbian forum https://forum.armbian.com/topic/6069-rpigpio-replacement-for-orange-pi-zero-and-nano-pi-duo and the reply. Also see https://forum.armbian.com/topic/5655-armbianio-api-proposal and https://www.cnx-software.com/2017/11/03/learn-more-about-linuxs-new-gpio-user-space-subsystem-libgpiod

Any plans to wrap libgpiod?

rm-hull, 2018-01-08:
No immediate plans; I suspect it will be quite some time before sysfs becomes deprecated.

Would be good to do some analysis to see how much effort is needed to support libgpiod though.

sgjava, 2018-01-08:
It was supposed to be deprecated as part of kernel 4.8 I think, but yes it's still there :) I wrote the Python wrappers for https://github.com/bitbank2/ArmbianIO (fast C based I2C, SPI and GPIO lib for all Armbian SBCs) and I believe Larry is going to port that to libgpiod at some point. I may end up doing a RPi.GPIO type wrapper, so it will have a wide range of SBCs supported instead of all the one off RPi.GPIO clones. It will make it easier to run Luma and other libs that use RPi.GPIO API.

closed 2018-01-08 by sgjava

sgjava, 2018-01-24:
I have libgpiod building on Armbian and should work on any mainline version https://github.com/sgjava/libgpiod-extra. libgpiod should perform better and has more reliable edge detection.