rm-hull/OPi.GPIO #23: IOError: [Errno 13] Permission denied: '/sys/class/gpio/gpio298/direction'

elpimous, 2018-01-21:
Ubuntu 16.04
Jetson TX2 board (aarch64, ARMv8)

Hello all.
Thanks to LUMA libs, my robot has eyes.

I need to run my python without root privileges, and it produces an error !

```
python /home/nvidia/catkin_ws/src/alfred_eyes/src/RandomOledEyes.py
Traceback (most recent call last):
  File "/home/nvidia/catkin_ws/src/alfred_eyes/src/RandomOledEyes.py", line 21, in <module>
    serial = spi(port=3, device=0, bus_speed_hz=8000000, transfer_size=2048, gpio=GPIO, gpio_DC=1, gpio_RST=2)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 253, in __init__
    bitbang.__init__(self, gpio, transfer_size, DC=gpio_DC, RST=gpio_RST)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 153, in __init__
    self._DC = self._configure(kwargs.get("DC"))
  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 164, in _configure
    self._gpio.setup(pin, self._gpio.OUT)
  File "/usr/local/lib/python2.7/dist-packages/OPi/GPIO.py", line 366, in setup
    sysfs.direction(pin, direction)
  File "/usr/local/lib/python2.7/dist-packages/OPi/sysfs.py", line 32, in direction
    with open(path, "w") as fp:

IOError: [Errno 13] Permission denied: '/sys/class/gpio/gpio298/direction'
```

Does anyone has idea to correct it ? Anything like udev ?
Thanks, Vincent.

elpimous, 2018-01-21:
here are specs : 
```
nvidia@tegra-ubuntu:~$ ls -lL /sys/class/gpio/gpio298
total 0
-rw-r--r-- 1 root root 4096 Jan 21 19:39 active_low
drwxr-xr-x 4 root root    0 Jan 21 15:01 device
-rw-r--r-- 1 root root 4096 Jan 21 19:28 direction
-rw-r--r-- 1 root root 4096 Jan 21 19:39 edge
drwxr-xr-x 2 root root    0 Jan 21 19:39 power
drwxrwx--- 2 root gpio    0 Jan 21 19:28 subsystem
-rw-r--r-- 1 root root 4096 Jan 21 19:28 uevent
-rw-r--r-- 1 root root 4096 Jan 21 19:39 value
```

sgjava, 2018-01-21:
This is for the NanoPi Duo, but maybe it will work with sysfs in general http://www.friendlyarm.com/Forum/viewtopic.php?f=59&t=949

rm-hull, 2018-01-22:
This issue was discussed in #14 - see also https://forum.armbian.com/topic/5225-gpio-from-python-user-space-opigpio-app/

elpimous, 2018-02-22:
Hi Richard.
Well, it doesn't work for me. Bad !