rm-hull/OPi.GPIO #24: OPi Zero internal pullup/down error

OperatorB, 2018-02-11:
Hi!
After a recent OS (dietpi) and firmware update on my OPi Zero, my python code stopped working. I found that the problem is around "GPIO.setup(..., pull_up_down=GPIO.PUD_UP)". 
When I run the code it spits out "AttributeError: 'module' object has no attribute 'PUD_UP' ".
Probably I screwed up something but I could not figure out what.

my code:

> #!/usr/bin/python
>import OPi.GPIO as GPIO
>GPIO.setmode(GPIO.BCM)
>GPIO.setup(11,GPIO.IN,pull_up_down=GPIO.PUD_UP)

when run with ./mycode.py it replies:

>Traceback (most recent call last):
>  File "./test.py", line 6, in <module>
>    GPIO.setup(11,GPIO.IN,pull_up_down=GPIO.PUD_UP)
>AttributeError: 'module' object has no attribute 'PUD_UP'
  
Thanks for reading ;)

rm-hull, 2018-02-12:
If you do `pip list | grep -i gpio` on the command-line what does it show?

OperatorB, 2018-02-12:
It replied:

DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
OPi.GPIO (0.3.1)

rm-hull, 2018-02-12:
Ok, so you have the latest version. But you know that pull-up/down is not yet supported? See #9

However GPIO.PUD_UP should exist.

What version of python are you using?

OperatorB, 2018-02-12:
All right, so there was no issue with it before the 'new' OPi firmware. Ok, thank you for the intel.

I'm using Python 2.7 (but 3.5 is installed as well). I will add externally a resistor then.

closed 2018-02-12 by OperatorB