rm-hull/OPi.GPIO #26: Rising Edge Detection innacurate

yaymuffins, 2018-05-01:
Hello,

I wanted to use a RPi script to use a rotary dialer, https://github.com/Szpeja/RotaryPi/blob/master/modules/RotaryDial.py

But it seem that rising edge detection is somewhat inaccurate, sometime detecting one less change in input than it should.

While I don't know much about Python at all, I tried a simple script to actively monitor the "button"

```
import OPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)

GPIO.setup(2, GPIO.IN)

while True:
    input_state = GPIO.input(2)
    if input_state == True:
        print('Button Pressed')
        time.sleep(0.01)
    else:
        print('nope')
        time.sleep(0.01)
```

I've ran both at the same time, and while the output of that small script was what I expected (10 "Button Pressed" for 10 impulsion)

```
...
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
nope
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
nope
nope
nope
nope
```

The output of the one depending on rising edge detection only detected 9 changes

```
[INPUT] 1 (2)
[INPUT] 0 (2)
[INPUT] 1 (2)
[INPUT] 0 (2)
[INPUT] 1 (2)
[INPUT] 1 (2)
[INPUT] 0 (2)
[INPUT] 1 (2)
[INPUT] 0 (2)
[INPUT] 1 (2)
[INPUT] 0 (2)
[INPUT] 1 (2)
[INPUT] 0 (2)
[INPUT] 1 (2)
[INPUT] 0 (2)
[INPUT] 1 (2)
[INPUT] 0 (2)
[INPUT] 1 (2)
[INPUT] 0 (2)
[DIGIT] Got digit: 9
[NUMBER] We have: 9
```

You can especially see at the line 5 and 6 that it didn't detect one of the change.

It get it right sometime, but not always, and it doesn't seem to be an hardware issue as monitoring the pin manually give me the correct number of value change, I don't know if it's fixable but the problem definitively seem to be around rising edge detection.

Race666, 2018-07-13:
Same problem here. When configuring, for example PIN26, to the rising edge:  It fires an Interrupt 2times for one change at Pin26. 
20180712 21:44:39 - TestGPIO - INFO - Started
20180712 21:45:18 - TestGPIO - DEBUG - Pulse 26 detected = 1
20180712 21:45:18 - TestGPIO - DEBUG - Pulse 26 detected = 2
20180712 21:46:46 - TestGPIO - DEBUG - Pulse 26 detected = 3
20180712 21:46:46 - TestGPIO - DEBUG - Pulse 26 detected = 4
20180712 21:48:15 - TestGPIO - DEBUG - Pulse 26 detected = 5
20180712 21:48:15 - TestGPIO - DEBUG - Pulse 26 detected = 6

But it seems it depends on the IO Port
20180712 21:50:48 - TestGPIO - INFO - Started
20180713 21:50:51 - TestGPIO - DEBUG - Pulse 12 detected = 1
20180714 21:51:03 - TestGPIO - DEBUG - Pulse 12 detected = 2
20180715 21:51:15 - TestGPIO - DEBUG - Pulse 12 detected = 3

Code:
```
#!/usr/bin/env python
# coding: utf-8
###############################################################################
import OPi.GPIO as GPIO
import time
import logging
from logging.handlers import SysLogHandler 
PulseCount=0

####### Pulse Tarif        
def PulseDetected(IOPort):
    global PulseCount
    global logger
    PulseCount+=1
    logger.debug("Pulse %d detected = %d",IOPort,PulseCount)
IOPin=26
logger = logging.getLogger("TestGPIO")
hConsoleLog = logging.StreamHandler()
hConsoleLog.setLevel(logging.DEBUG)
hConsoleLog.setFormatter(logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s',"%Y%m%d %H:%M:%S"))
logger.addHandler(hConsoleLog)

logger.setLevel(logging.DEBUG)
GPIO.setmode(GPIO.BOARD)
GPIO.setup(IOPin, GPIO.IN, pull_up_down=0)	
GPIO.add_event_detect(IOPin, GPIO.RISING, callback=PulseDetected)
logger.info('Started')           
while True:
    time.sleep(0.1)
```

rm-hull, 2019-01-29:
I think this probably needs some form of debounce behaviour to make it work as you expect. There is an outstanding task for that, see issue #10

closed 2019-01-29 by rm-hull