rm-hull/OPi.GPIO #34: Has anyone got GPIO input working on OrangePi Prime?

mikalstill, 2019-01-12:
I for the life of me can't get it to work. Here's the code I am using:

```
import OPi.GPIO as GPIO
import orangepi.pc
import time

def pin_number(letter, digit):
    return (ord(letter) - ord('A')) * 32 + digit

pin = pin_number('A', 13)

GPIO.setmode(orangepi.pc.BOARD)
GPIO.setwarnings(False)
GPIO.setup(pin, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)

def event_callback(channel):
    print('Event detected: %s' % GPIO.input(pin))
    
GPIO.add_event_detect(pin, GPIO.BOTH, callback=event_callback, bouncetime=50)

while True:
    time.sleep(1)
```

I simply never get an edge detected. Output works just fine.

mikalstill, 2019-01-14:
Ok, so I think I've found a timing issue. Here's a version of the above code which works correctly (noting that I also changed pin numbering scheme in an unrelated change and that this code isn't using edge detection):

```
import OPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.SUNXI)
GPIO.setwarnings(False)
GPIO.setup('PA7', GPIO.IN, pull_up_down=GPIO.PUD_DOWN)

time.sleep(1)
with open('/sys/class/gpio/gpio7/direction', 'w') as f:
    f.write('in')

while True:
    print('Reading...')
    if GPIO.input('PA7') == GPIO.LOW:
        print('Pressed')
    else:
        print('Released')
    time.sleep(1)
```

I'm going to go spelunking and try and work out what the timing issue is.

rm-hull, 2019-01-14:
Maybe related to #26 ?

mikalstill, 2019-01-15:
You know, I think the overwhelming evidence at this point is that my failure was user error. I now have this working and have written up what I did at http://www.madebymikal.com/updated-examples-for-orangepi-gpios/ so that I can remember it in the future.

Sorry for wasting your time with this one.

closed 2019-01-15 by mikalstill

Hazeline2018, 2019-01-29:
Hi Mikal, I tried exactly your sample code, blinking the LED with the proper pin numbering to my board, but got the following error. No SUNXI attribute. What am I missing? thanks!

  File "gpio_test.py", line 11, in <module>
    GPIO.setmode(GPIO.SUNXI)
AttributeError: module 'OPi.GPIO' has no attribute 'SUNXI'