rm-hull/OPi.GPIO #37: Is OPi Lite 2 being support yet?

Hazeline2018, 2019-01-26:
Hi, I'm working on OPi Lite 2 board and came across this page for porting RPi.GPIO lib into OPi board. Is anyone working on getting Lite 2 board supported by this drop-in lib yet? If yes, I might want to give it a try.
thanks,!

rm-hull, 2019-01-29:
If you can supply the correct mappings for the OPi Lite 2 SBC, then please do so. See one of the existing PRs for guidance on what you need to do, e.g.: https://github.com/rm-hull/OPi.GPIO/pull/30/files

Hazeline2018, 2019-01-29:
okay, once I test out working, will upload

Hazeline2018, 2019-01-29:
I've created this pin mapping table for lite2 board. But still confused about this BCM mapping. I understand that's the actual GPIO pin mapping to the actual BCM SoC pins. However, for OPi, it's not using BRCM SoC, instead, Allwinner H* chips. How should I map GPIO pins to BCM? If I'm just going to use Board mode or even this SUNXI mapping mode, can I just ignore BCM mapping?

Hazeline2018, 2019-01-29:
I also realized that if I'm to use SUNXI mapping, which is really just a simple calc based on the GPIO name and much less confusing IMO, there is no need to create such a board or BCM mapping table anymore because all H* chips GPIO follow the same formula. And I can just simple install this module and use as is. Is this really true?