rm-hull/OPi.GPIO #40: [feature request] Add pwm support

pedrohoffmann, 2019-02-05:
Hello,
Do you plan to add Pwm support the the library?

rm-hull, 2019-02-05:
I have no plans to add i myself, but PRs welcome

pedrohoffmann, 2019-02-05:
No problem, im a noob on dev, but I'll take a look and try :)
great job on this btw

enesbcs, 2020-04-15:
There is a simple SoftPWM (not so fast) implementation which can be used with OPi.GPIO by changing:

```
from pyA20.gpio import gpio as GPIO
from pyA20.gpio import port
```
to

`import OPi.GPIO as GPIO`

gounthar, 2021-11-18:
There is a `Demo/PWM_Demo.py` that does nothing on my machine (orangepizero, OS: Debian 11 bullseye, Kernel: armv7l Linux 5.10.60-sunxi), even if I manage to get a LED blinking with a simple script (with pin 7, BOARD mode).
In the PWM_Demo.py file, the PWM_pin is listed as `0`. When I use `7` as for the simple LED blink, I get this error:

```bash
OSError: [Errno 19] No such device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/OPi.GPIO/Demo/PWM_Demo.py", line 10, in <module>
    p = GPIO.PWM(PWM_chip, PWM_pin, frequency_Hz, Duty_Cycle_Percent)    # new PWM on channel=LED_gpio frequency=38KHz
  File "/usr/local/lib/python3.9/dist-packages/OPi.GPIO-0.5.2-py3.9.egg/OPi/GPIO.py", line 743, in __init__
    raise e
  File "/usr/local/lib/python3.9/dist-packages/OPi.GPIO-0.5.2-py3.9.egg/OPi/GPIO.py", line 729, in __init__
    sysfs.PWM_Export(chip, pin)  # creates the pwm sysfs object
  File "/usr/local/lib/python3.9/dist-packages/OPi.GPIO-0.5.2-py3.9.egg/OPi/sysfs.py", line 95, in PWM_Export
    fp.write(str(pin))
OSError: [Errno 19] No such device
```
In `armbian-config`, I selected `pwm` and `w1-gpio`.

Do you know what I did wrong?

Thanks.

Fionnoch, 2022-08-05:
I think you're changing the pwm chip not the pin. Try leaving the chip as 0 and see what happens, if it gives an error saying the device doesn't exist try change it to 1, you shouldn't go above 2. Note as well that the chip is mapped to a specific pin and that pin may not be whats listed as the pwm pin on the pinout diagrams for the board (its a pain I know). For example in the image below I pointed where the pwm pin is located on that board. 

Hope this helps!

[image]