rm-hull/OPi.GPIO #47: usermod: group 'gpio' does not exist

gounthar, 2020-04-10:
Hi,

I just installed OPi.GPIO on my OrangePi Zero running Armbian buster with Linux 5.4.31-sunxi.
I followed the [documentation](https://opi-gpio.readthedocs.io/en/latest/install.html) but got the following error:

```bash
sudo pip3 install --upgrade OPi.GPIO
Collecting OPi.GPIO
  Downloading https://files.pythonhosted.org/packages/32/88/be49fa348875993d033cd4234ff4f27fab97265fe3e2687de3009816190d/OPi.GPIO-0.4.0-py2.py3-none-any.whl
Installing collected packages: OPi.GPIO
Successfully installed OPi.GPIO-0.4.0
sudo usermod -aG gpio poddingue
usermod: group 'gpio' does not exist
```

Do I have to create the group gpio by myself, or should the pip3 install do it for me?
Thanks.

rm-hull, 2020-04-10:
You should create it yourself. Pip3 won't do it for you. I am surprised the group does not already exist in the os though.

gounthar, 2020-04-10:
I will report to the distro maintainers, thanks.

closed 2020-04-10 by gounthar