rm-hull/OPi.GPIO #49: OrangePi One Plus problem with Interrupts and Edge detection

carlosarmentac, 2020-09-09:
I have an issue with events detection in my [OrangePi One Plus](http://www.orangepi.org/OrangePiOneplus/)  . I'm running this script as a root user:

```python
import orangepi.oneplus
import OPi.GPIO as GPIO
from time import sleep

channel = 21

GPIO.setmode(orangepi.oneplus.BOARD)
GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)

def my_callback(channel):
    print('This is a edge event callback function!')
    print('Edge detected on channel %s'%channel)
    print('This is run in a different thread to your main program')

GPIO.add_event_detect(channel, GPIO.RISING, callback=my_callback)

while True:    
    sleep(.5)

GPIO.cleanup()
print("Bye")
```

The output is:

```bash
inputsInterruptions.py:8: UserWarning: Pull up/down setting are not (yet) fully supported, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)
inputsInterruptions.py:8: UserWarning: Channel 21 is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 74, in run
    sysfs.edge(self._pin, NONE)
  File "/usr/local/lib/python2.7/dist-packages/OPi/sysfs.py", line 84, in edge
    with open(path, "w") as fp:
IOError: [Errno 13] Permission denied: '/sys/class/gpio/gpio67/edge'

Traceback (most recent call last):
  File "inputsInterruptions.py", line 20, in <module>
    GPIO.cleanup()
  File "/usr/local/lib/python2.7/dist-packages/OPi/GPIO.py", line 572, in cleanup
    cleanup(list(_exports.keys()))
  File "/usr/local/lib/python2.7/dist-packages/OPi/GPIO.py", line 578, in cleanup
    cleanup(ch)
  File "/usr/local/lib/python2.7/dist-packages/OPi/GPIO.py", line 582, in cleanup
    event.cleanup(pin)
  File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 162, in cleanup
    remove_edge_detect(pin)
  File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 144, in remove_edge_detect
    _threads[pin].cancel()
  File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 45, in cancel
    self.join()
  File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 81, in join
    raise e
IOError: [Errno 13] Permission denied: '/sys/class/gpio/gpio67/edge'
```

gpio readall output:
```bash
$ gpio readall
 +------+-----+----------+------+---+  OPi H6  +---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 |      |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |      |
 |  230 |   0 |    SDA.1 | ALT4 | 0 |  3 || 4  |   |      | 5V       |     |      |
 |  229 |   1 |    SCL.1 | ALT4 | 0 |  5 || 6  |   |      | GND      |     |      |
 |  228 |   2 |     PWM1 |  OFF | 0 |  7 || 8  | 0 | OFF  | PD21     | 3   | 117  |
 |      |     |      GND |      |   |  9 || 10 | 0 | OFF  | PD22     | 4   | 118  |
 |  120 |   5 |    RXD.3 | ALT2 | 0 | 11 || 12 | 0 | OFF  | PC09     | 6   | 73   |
 |  119 |   7 |    TXD.3 | ALT2 | 0 | 13 || 14 |   |      | GND      |     |      |
 |  122 |   8 |    CTS.3 |  OUT | 0 | 15 || 16 | 0 | IN   | PC08     | 9   | 72   |
 |      |     |     3.3V |      |   | 17 || 18 | 0 | OFF  | PC07     | 10  | 71   |
 |   66 |  11 |   MOSI.0 |  OFF | 0 | 19 || 20 |   |      | GND      |     |      |
 |   67 |  12 |   MISO.0 |   IN | 1 | 21 || 22 | 0 | ALT2 | RTS.3    | 13  | 121  |
 |   64 |  14 |   SCLK.0 |  OUT | 0 | 23 || 24 | 0 | OFF  | CE.0     | 15  | 69   |
 |      |     |      GND |      |   | 25 || 26 | 0 | OFF  | PH03     | 16  | 227  |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+  OPi H6  +---+------+----------+-----+------+
```

what i'm doing wrong?

Actually I have success with inputs polling method and controlling outputs.

Thank you in advance!!!

yuzi40277738, 2021-02-20:
My test is OK！

oot@huangarm:/home/huang/test-1# python3 led_s-3.py
led_s-3.py:21: UserWarning: Pull up/down setting are not (yet) fully supported, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
led_s-3.py:21: UserWarning: Channel 7 is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C^[[C
^CTraceback (most recent call last):
  File "led_s-3.py", line 31, in <module>
    time.sleep (0.5)
KeyboardInterrupt

root@huangarm:/home/huang/test-1# python3 led_s-3.py
led_s-3.py:21: UserWarning: Pull up/down setting are not (yet) fully supported, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
led_s-3.py:21: UserWarning: Channel 7 is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7
This is run in a different thread to your main program
This is a edge event callback function!
Edge detected on channel 7

Fionnoch, 2021-04-22:
Try run as root, this library uses sysfs which can sometimes create permission issues, if you need to run this as a non root user you can great an access group and add the sysfs folder that is denying you access to it i.e '/sys/class/gpio/gpio67/edge'. 

In your case if you gave yourself access to the '/sys/class/gpio/gpio67/' folder that should give you access to all the function files within it.

fabricio-acti, 2022-08-03:
> I have an issue with events detection in my [OrangePi One Plus](http://www.orangepi.org/OrangePiOneplus/) . I'm running this script as a root user:
> 
> ```python
> import orangepi.oneplus
> import OPi.GPIO as GPIO
> from time import sleep
> 
> channel = 21
> 
> GPIO.setmode(orangepi.oneplus.BOARD)
> GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)
> 
> def my_callback(channel):
>     print('This is a edge event callback function!')
>     print('Edge detected on channel %s'%channel)
>     print('This is run in a different thread to your main program')
> 
> GPIO.add_event_detect(channel, GPIO.RISING, callback=my_callback)
> 
> while True:    
>     sleep(.5)
> 
> GPIO.cleanup()
> print("Bye")
> ```
> 
> The output is:
> 
> ```shell
> inputsInterruptions.py:8: UserWarning: Pull up/down setting are not (yet) fully supported, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
>   GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)
> inputsInterruptions.py:8: UserWarning: Channel 21 is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
>   GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)
> Exception in thread Thread-1:
> Traceback (most recent call last):
>   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
>     self.run()
>   File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 74, in run
>     sysfs.edge(self._pin, NONE)
>   File "/usr/local/lib/python2.7/dist-packages/OPi/sysfs.py", line 84, in edge
>     with open(path, "w") as fp:
> IOError: [Errno 13] Permission denied: '/sys/class/gpio/gpio67/edge'
> 
> Traceback (most recent call last):
>   File "inputsInterruptions.py", line 20, in <module>
>     GPIO.cleanup()
>   File "/usr/local/lib/python2.7/dist-packages/OPi/GPIO.py", line 572, in cleanup
>     cleanup(list(_exports.keys()))
>   File "/usr/local/lib/python2.7/dist-packages/OPi/GPIO.py", line 578, in cleanup
>     cleanup(ch)
>   File "/usr/local/lib/python2.7/dist-packages/OPi/GPIO.py", line 582, in cleanup
>     event.cleanup(pin)
>   File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 162, in cleanup
>     remove_edge_detect(pin)
>   File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 144, in remove_edge_detect
>     _threads[pin].cancel()
>   File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 45, in cancel
>     self.join()
>   File "/usr/local/lib/python2.7/dist-packages/OPi/event.py", line 81, in join
>     raise e
> IOError: [Errno 13] Permission denied: '/sys/class/gpio/gpio67/edge'
> ```
> 
> gpio readall output:
> 
> ```shell
> $ gpio readall
>  +------+-----+----------+------+---+  OPi H6  +---+------+----------+-----+------+
>  | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
>  +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
>  |      |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |      |
>  |  230 |   0 |    SDA.1 | ALT4 | 0 |  3 || 4  |   |      | 5V       |     |      |
>  |  229 |   1 |    SCL.1 | ALT4 | 0 |  5 || 6  |   |      | GND      |     |      |
>  |  228 |   2 |     PWM1 |  OFF | 0 |  7 || 8  | 0 | OFF  | PD21     | 3   | 117  |
>  |      |     |      GND |      |   |  9 || 10 | 0 | OFF  | PD22     | 4   | 118  |
>  |  120 |   5 |    RXD.3 | ALT2 | 0 | 11 || 12 | 0 | OFF  | PC09     | 6   | 73   |
>  |  119 |   7 |    TXD.3 | ALT2 | 0 | 13 || 14 |   |      | GND      |     |      |
>  |  122 |   8 |    CTS.3 |  OUT | 0 | 15 || 16 | 0 | IN   | PC08     | 9   | 72   |
>  |      |     |     3.3V |      |   | 17 || 18 | 0 | OFF  | PC07     | 10  | 71   |
>  |   66 |  11 |   MOSI.0 |  OFF | 0 | 19 || 20 |   |      | GND      |     |      |
>  |   67 |  12 |   MISO.0 |   IN | 1 | 21 || 22 | 0 | ALT2 | RTS.3    | 13  | 121  |
>  |   64 |  14 |   SCLK.0 |  OUT | 0 | 23 || 24 | 0 | OFF  | CE.0     | 15  | 69   |
>  |      |     |      GND |      |   | 25 || 26 | 0 | OFF  | PH03     | 16  | 227  |
>  +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
>  | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
>  +------+-----+----------+------+---+  OPi H6  +---+------+----------+-----+------+
> ```
> 
> what i'm doing wrong?
> 
> Actually I have success with inputs polling method and controlling outputs.
> 
> Thank you in advance!!!

Conseguiu achar a solução?
Estou com o mesmo problema no Orange PI 3 LTS