rm-hull/OPi.GPIO #55: orangepi4 error ?

Shaggy013, 2021-06-02:
writing byte-compilation script '/var/tmp/portage/dev-python/OPi-GPIO-0.5.0/temp/tmpdqhcd8x1.py'
/usr/bin/python3.8 /var/tmp/portage/dev-python/OPi-GPIO-0.5.0/temp/tmpdqhcd8x1.py
  File "/usr/lib/python3.8/site-packages/orangepi/pi4B.py", line 21
    import orangepi.4
                             ^
SyntaxError: invalid syntax

now i wonder why it faults there ?

rm-hull, 2021-06-02:
@drizzle1 can you help? looking at the history, this was added in PR #52

Shaggy013, 2021-06-02:
ow the  ^ at the . between orangepi and 4 the other files dont give a warning or 4 is to short or it is not happy about it is a number instead of a name ?

Shaggy013, 2021-06-03:
ah he saw it ;) 

solved

rm-hull, 2021-06-03:
Fixed in #56

closed 2021-06-03 by rm-hull