rm-hull/OPi.GPIO #62: Does it support orange pi 3lts？

KKmmmin, 2022-03-21:


timo12357, 2022-08-11:
I have an Orangepi 3 lts and I have not been able to make it work. I guess it indeed does NOT work with Orangepi 3 lts.

mjeveritt, 2022-10-24:
It may need a new board config file creating ..

timo12357, 2022-10-24:
How do I do that?

ggdxwz, 2023-05-20:
Tested, does not work on pi3 lts.

ggdxwz, 2023-05-20:
To be specific, this is the error I am getting:
`  File "test.py", line 3
    import orangepi.3
                             ^^
SyntaxError: invalid syntax`
Not sure if it is an user error or something wrong with my environment / installation

pdrlmfr, 2023-08-02:
@ggdxwz have you solved? i got the same error

nathan-osman, 2023-10-04:
I get:

```
>>> import OPi.GPIO as GPIO
>>> GPIO.setmode(GPIO.BOARD)
>>> GPIO.setup(12, GPIO.OUT)
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "[...]/lib/python3.11/site-packages/OPi/GPIO.py", line 482, in setup
    sysfs.direction(pin, direction)
  File "[...]/lib/python3.11/site-packages/OPi/sysfs.py", line 52, in direction
    with open(path, "w") as fp:
OSError: [Errno 22] Invalid argument
```