rm-hull/OPi.GPIO #65: usermod: group 'gpio' does not exist

NadavK, 2022-07-15:
Hardware: OrangePi PC (H3)
OS: Armbian 22.05.3 Jammy with Linux 5.15.48-sunxi
Issue: running ```sudo usermod -aG gpio``` produces error: ```usermod: group "gpio" "does not exist"```

I created the group manually (per https://github.com/rm-hull/OPi.GPIO/issues/47), and continued to follow the instructions but am still not able to access GPIO with non-root user (```Permission denied  '/sys/class/gpio/gpio7/direction'```).

Can it be there is a comma error in the instructions? 
The instruction use "[Right Double Quotation Mark](https://www.compart.com/en/unicode/U+201D)", "[Left Double Quotation Mark](https://www.compart.com/en/unicode/U+201C)" and "[Left Single Quotation Mark](https://www.compart.com/en/unicode/U+2018)" instead of the more common "[Quotation Mark](https://www.compart.com/en/unicode/U+0022)" and "[Apostrophe](https://www.compart.com/en/unicode/U+0027)"
I replaced the above with the common chars, but am still not able to obtain non-root access:
```SUBSYSTEM=="gpio", KERNEL=="gpiochip*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:gpio /sys/class/gpio/export /sys/class/gpio/unexport ; chmod 220 /sys/class/gpio/export /sys/class/gpio/unexport'" SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:gpio /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value ; chmod 660 /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value'"```

drawpaint, 2022-07-21:
Same issue here. So I just run my program as root.

varna9000, 2022-10-02:
same here. Anyone found the problem?

varna9000, 2022-10-02:
Ok, just dug a bit more in here. Apparently the docs hasn't been updated. According to the instruction [here](doc/install.rst) udev rule lines need to be swapped. In the [docs](https://opi-gpio.readthedocs.io/en/latest/install.html) it's in the wrong order.

mjeveritt, 2022-10-24:
Looking at my O-Pi there definitely isn't a `gpio` group in `/etc/groups` .. I am somewhat doubtful that udev creates a group too.

A quick google leads me to https://forum.armbian.com/topic/1886-gpio-access-from-user-space/ which will probably help with the non-root access; you're right - the documentation really should be updated .. I'm sure PR's would be welcome! :grin: 

> Ok, just dug a bit more in here. Apparently the docs haven't been updated. According to the instruction [here](https://github.com/rm-hull/OPi.GPIO/blob/master/doc/install.rst) udev rule lines need to be swapped. In the [docs](https://opi-gpio.readthedocs.io/en/latest/install.html) the rules are in the wrong order.

Was this not fixed in https://github.com/rm-hull/OPi.GPIO/pull/41 ?

The Sphinx docs are generated from the RST in the repo, so technically both should match (unless one needs rebuilding?!)
@rm-hull ?

DansDesigns, 2023-12-25:
dont know if its been fixed as i just encountered this issue today on a NanoPi Neo v1.4 running armbian bookworm 6.1.63

medeiros1, 2024-02-09:
I still have the problem with the latest bookworm armbian on an OrangePi PC. I tried adding the udev rules, but it didn't help for me.  I am using the WiringOP library and get the following Message from my application when I start as normal user: 
----------------------
wiringPiSetup: Unable to open /dev/mem or /dev/gpiomem: Datei oder Verzeichnis nicht gefunden.
  Aborting your program because if it can not access the GPIO
  hardware then it most certianly won't work
  Try running with sudo? (with sudo it works, but that's not what I want)
----------------------
rolf@orangepipc:/sys/class/gpio$ ls -l
insgesamt 0
--w--w---- 1 root gpio 4096  9. Feb 13:37 export
lrwxrwxrwx 1 root root    0  9. Feb 13:37 gpiochip0 -> ../../devices/platform/soc/1c20800.pinctrl/gpio/gpiochip0
lrwxrwxrwx 1 root root    0  9. Feb 13:37 gpiochip352 -> ../../devices/platform/soc/1f02c00.pinctrl/gpio/gpiochip352
--w--w---- 1 root gpio 4096  9. Feb 13:37 unexport

rolf@orangepipc:~$ cat /etc/group |grep rolf
sys:x:3:rolf
tty:x:5:rolf
disk:x:6:rolf
kmem:x:15:rolf
dialout:x:20:rolf
sudo:x:27:rolf
audio:x:29:rolf
video:x:44:rolf
plugdev:x:46:rolf
games:x:60:rolf
users:x:100:rolf
systemd-journal:x:999:rolf
input:x:102:rolf
i2c:x:108:rolf
netdev:x:113:rolf
rolf:x:1000:
gpio:x:1001:rolf

I would highly appreciate to find a proper solution

Thanks in advance, Rolf