rm-hull/OPi.GPIO #73: orangepi zero2 trouble issue help

cloudy4125, 2023-02-18:
orangepi zero2 python3 

import time
import OPi.GPIO as GPIO
GPIO.setmode(GPIO.BOARD)
not working 

GPIO.setmode(GPIO.orangepi.zero2)
not working

import orangepi.zero2
   from OPi import GPIO
   GPIO.setmode(orangepi.zero2.BOARD)
not working

GPIO.setboard(GPIO.ZERO2)
GPIO.setmode(GPIO.BOARD)
not working

how to working in  GPIO.setboard

I want to know how to turn off or turn on the machine.

how to serboard, setmode

I wonder how to set it up.

soimy, 2023-02-21:
You need to 
```python
import orangepi.zero2
import OPi.GPIO as gio
gio.setmode(orangepi.zero2.BOARD)

cloudy4125, 2023-02-22:
> Try
> 
> ```python
> import orangepi.zero2
> import OPi.GPIO as gpio
> gpio.setmode(orangepi.zero2.BOARD)
> ```
> 
> working for me Note that only BOARD pin is supported, ref: #64

Installation process, access permission, but it comes out like this.

Traceback (most recent call last):
  File "/home/orangepi/button.py", line 1, in <module>
    import orangepi.zero2
ModuleNotFoundError: No module named 'orangepi.zero2'

soimy, 2023-02-22:
PR #64 is after the released v0.5.2, so you have to install the GitHub version instead of pypi one.
```
git+https://github.com/rm-hull/OPi.GPIO@f1794fcb1b2f78189b9619633d8680a6a213686f#egg=opi-gpio
```

rm-hull, 2023-02-22:
ill see about publishing later today

cloudy4125, 2023-03-01:
> PR #64 is after the released v0.5.2, so you have to install the GitHub version instead of pypi one.
> 
> ```
> git+https://github.com/rm-hull/OPi.GPIO#egg=opi-gpio
> ```

Thank you. It's working.

closed 2023-03-01 by cloudy4125