rm-hull/OPi.GPIO #79: Add support for Orange Pi Zero 3

PicoPlanetDev, 2023-08-03:
The Orange Pi Zero 3 uses an Allwinner H618 processor which appears to be very similar to the H616 used in the Zero 2. In fact, the pin mappings may even be identical. Here's a link to the Zero 3 schematic: [OrangePi-Schematic of ZERO3.pdf](https://drive.google.com/file/d/1GelRJz-6Dg4i_EQ1SwrfHeqfLOPw9kdO/view)

As soon as I get my Zero 3 in the mail, I'll see if I can figure out the physical to GPIO pin mappings and create a pull request. In the meantime, I'd love to hear if anyone has experimented with it yet!

BorretSquared, 2023-10-17:
Would kill for this.

PicoPlanetDev, 2023-10-17:
I submitted a pull request but no news on it for a few months. In the meantime, I've got my fork https://github.com/PicoPlanetDev/OPi.GPIO that simply copies the Zero2 mapping to a new Zero3 file. This has been working flawlessly on my own Zero3 so far. Maybe this helps, although let me know if you want me to flesh it out a little further if needed.

techsd, 2023-11-01:
# pin number = (position of letter in alphabet - 1) * 32 + pin number
# So, PD14 will be (4 - 1) * 32 + 14 = 110

# Orange Pi Zero 3 physical board pin to GPIO pin
BOARD = {
    3:    229,   # PH5/I2C3_SDA
    5:    228,   # PH4/I2C3_SCK
    7:    73,    # PC9
    8:    226,   # PH2/UART5_TX
    10:   227,   # PH3/UART5_RX
    11:   70,    # PC6
    12:   75,    # PC11
    13:   69,    # PC5
    15:   72,    # PC8
    16:   79,    # PC15
    18:   78,    # PC14
    19:   231,   # PH7,SPI1_MOSI
    21:   232,   # PH8,SPI1_MISO
    22:   71,    # PC7
    23:   230,   # PH6,SPI1_CLK
    24:   233,   # PH9,SPI1_CS
    26:   74,    # PC10
    27:   234,   # PH10/UART6_TX
    28:   235,   # PH11/UART6_RX
    29:   236,   # PH12/I2C4_SDA
    30:   237,   # PH13/I2C4_SCK
    31:   238,   # PH14/SPI2_MOSI
    33:   239,   # PH15/SPI2_MISO
    34:   240,   # PH16/SPI2_CLK
    35:   241,   # PH17/SPI2_CS
    36:   242,   # PH18/UART7_TX
    37:   243,   # PH19/UART7_RX
}

# BCM mapping not identified yet, keeping it for compatibility
BCM = BOARD

JMVS, 2024-06-24:
Is this helpful at all?

[image]

[image]

[image]

Taken from: http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_Zero_3#26_Pin_Interface_Pin_Description

JMVS, 2025-09-30:
Is there any news on this? Are there any plans to implement support? I'm no coding expert, but if I can help, let me know.